pythops/bluetui #3: Center the UI

orhun, 2024-03-02:
On my terminal `bluetui` looks like this:

[image]

I think it would be nice to vertically center the widgets to make things more symmetric.

What do you think?

pythops, 2024-03-02:
I agree that it would be better with some symmetric layout. 
What layout do you suggest knowing that there is another block that appears when scanning for a new devices ?

orhun, 2024-03-02:
You can shift everything upwards when there is a new block and keep everything centered 🐻

pythops, 2024-03-02:
If you can send a PR for it, sure :)

orhun, 2024-03-05:
I made the changes:

[image]

What do you think of this? It still feels a bit odd to me. I'm wondering what's missing.

orhun, 2024-03-05:
I think fixed percentages are better (%33 for each block):

[image]

pythops, 2024-03-05:
Yeah I like the last one with fixed percentages better.

PR #5 (2024-03-06): Update the main layout constraints
fixes #3
closed 2024-03-06 by pythops