pythops/bluetui #12: Is it possible to advertise using the GUI?

potto216, 2024-03-31:
Is it possible to have the device controller go into advertise mode (not for pairing) to advertise information such as manufacturing data or custom data such as a sensor temperature, without having to setup a connection with a GATT database? Or is the focus of this program on paired connections?

pythops, 2024-03-31:
Sure, that would nice to have

potto216, 2024-03-31:
Okay if helpful, this week I can add some use cases to the issue for requirements development?

pythops, 2024-03-31:
Sure, do it

potto216, 2024-04-06:
Well I'm still thinking about use cases. A limitation to my idea of having bluetui control advertisers is that normally users would either want to advertise a very specific static piece of information such as a service uuid or would want the advertising data to be dependent on something like changing sensor data. Either way that would probably be better solved by a command line program which I discuss at https://github.com/bluez/bluer/issues/140.

A potentially useful need would be to organize advertising reports from different devices so they can be easily searchable or have alerts when certain values change such as RSSI. 

Right now I cannot think of a specific need, but am still investigating areas such as Periodic Advertising with Responses.

closed 2026-10-07 by pythops