pythops/bluetui #13: Feature suggestion GATT server/client support

potto216, 2024-03-31:
I think a great feature would be GATT server/client support for the software (assuming it is in the scope of the program's goals). Having a GUI based GATT client would be easier to understand a GATT server than using a text-based system for interaction.  Also having a GUI based GATT server would make it easy for someone to setup a GATT server than doing it programmatically or with a text-based program like bluetoothctl.

pythops, 2024-03-31:
Great feature to add for sure :100:

potto216, 2024-03-31:
Awesome, I can add some possible use cases for discussion and requirements development this week.

pythops, 2024-03-31:
Perfect, do it

potto216, 2024-04-06:
Something I have a need for and I believe others do is a GUI to navigate a GATT server database from a client. For example I have a simple GATT server on an arduino https://github.com/potto216/arduino-examples/tree/main/bluetooth/ble_gatt_server_multi_service with timer services and hardware pin control by the GATT client. Using bluetoothctl is a pain, and although I have a web bluetooth app it requires overhead to spin up. Having a GUI that a client can navigate a GATT database and modify values and set alerts when values change would be a great debugging and discovery tool.
@pythops what are your thoughts?

pythops, 2024-07-08:
working on it

potto216, 2024-07-10:
Oh great! I can help with the testing, documentation etc.

closed 2026-10-07 by pythops