pythops/bluetui #14: Unable to find advertising device when scanning to pair

potto216, 2024-03-31:
When scanning for advertising devices to pair with I am unable to see the advertising device. I am scanning for E4:5F:01:5E:2B:A6 which is advertising over the air and I don't see it in the scanning list shown below:
[image]

But on the same device if I exit `bluetui` use `bluetoothctl` I can see E4:5F:01:5E:2B:A6 (underlined in red) and it has strong signal strength. 
[image]

What should I do to see the advertising device? Thanks!

pythops, 2024-03-31:
:thinking:  
can you try those steps:
1.  try to emove the device using `bluetoothctl` cli
2. restart `bluetoothd` daemon
3. Put the device pairing mode
4. Use `bluetui` to scan for new devices

pythops, 2025-05-10:
@potto216 any update ?

potto216, 2025-06-09:
I'm closing the issue because I was not able to reproduce it and able to see an advertising device and connect to it (below)

[image]

closed 2025-06-09 by potto216