pythops/bluetui #18: The ✅︎ (tick mark) visible in the demo image not rendering properly.

mbilyanov, 2024-06-10:
Hi,

I just installed `bluetui` on Arch Linux and it is working great. Only problem is that the `tick-mark` (also visible in your demo inside the readme section of the project) is rendering as some garbage character. I have tested the console and it seems to support the unicode characters (emojis such as skulls etc.). Any ideas?

Kernel: `6.6.32-1-lts`
Bash: `5.2.26`
Installed through `paru -S bluetui`

pythops, 2024-06-10:
You need to install [nerdfonts](https://www.nerdfonts.com/#home). Try to install these packages
```
ttf-nerd-fonts-symbols
ttf-nerd-fonts-symbols-common
ttf-nerd-fonts-symbols-mono
```
let me know if that fix it for you

mbilyanov, 2024-06-10:
That worked. Thanks!

pythops, 2024-06-10:
Added a note in the Readme

closed 2024-06-10 by pythops