pythops/bluetui #19: Package for NixOS

sakshatshinde, 2024-07-01:
Consider packaging this for NixOS https://pavluk.org/blog/2023/05/30/nix_packaging.html

pythops, 2024-07-01:
I can look at it later, but you can send a PR meanwhile, that would be great :pray:

sakshatshinde, 2024-07-03:
I've never packaged any software (might take a shot at it) in the meantime ~ if anyone wants they can have a go at it https://nix.dev/tutorials/packaging-existing-software.html

Samuel-Martineau, 2024-07-26:
@sakshatshinde Is #21 what you had in mind?

alexandru0-dev, 2024-08-02:
@Samuel-Martineau great job, a flake is really good.
In the future can also be upstream as only some little modifications are needed

Samuel-Martineau, 2024-08-02:
@alexandru0-dev I was thinking of submitting a nixpkgs PR in the the next few weeks. I started with a flake because adding new packages to nixpkgs can be a very slow process.

alexandru0-dev, 2024-08-02:
@Samuel-Martineau ik, fully agree.
Also a flake let's you be at git version vs a release which nix (most of the time) uses

closed 2024-09-01 by pythops

donovanglover, 2024-09-10:
@Samuel-Martineau Thanks for packaging this! If you make a PR I can help merge bluetui into nixpkgs. Make sure to follow the format in the [nixpkgs manual](https://nixos.org/manual/nixpkgs/unstable/#compiling-rust-applications-with-cargo).

- Set `meta.mainProgram` to `"bluetui"`
- Replace `repo = pname;` with `repo = "bluetui";`
- Set `meta.platforms` to `lib.platforms.linux`

ArtemChandragupta, 2024-10-13:
Hi. How is the nixpkg progress?

donovanglover, 2024-10-16:
@ArtemChandragupta I made a PR here: https://github.com/NixOS/nixpkgs/pull/348939

commit ced9f86a (2024-09-01): Package for NixOS (#21)

Fix #19