pythops/bluetui #23: Feature request

LionyxML, 2024-07-30:
Thanks for this awesome package!

Is there a way to configure some bluetooth settings specific for the device? 

Like, on my headphones on gui (blueman-manager) I can set the mode it is working, like AD2P, HSP/HFP and also choose the codecs.

pythops, 2024-07-30:
I personally do not know that deep in the Bluetooth world but maybe this is the opportunity to do it :sweat_smile: 
I can look at for sure. Meanwhile, if someone out there knows already how to do it, plz go for it.

treeshateorcs, 2024-09-04:
i use pulsemixer for that

LionyxML, 2024-09-14:
Thanks @treeshateorcs!

This actually solves the problem of having "bluetooth and sound controls" all on CLI.

@pythops: As a log note for further reference:

On pulsemixer use F3, select the bluetooth device, hit enter:

[image]

Select the profile:

[image]

And that's it! :)

closed 2024-09-14 by LionyxML