pythops/bluetui #24: nothing is shown after installing "bluetui"

evanescente-ondine, 2024-07-30:
Hi.
I installed bluetui with cargo, and I see literally nothing, a blank terminal window.

[image]

Linux 84-47-09-37-40-35 6.9.11-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.9.11-1 (2024-07-25) x86_64 GNU/Linux

pythops, 2024-07-30:
can you check if `bluetooth` service is running.
```
systemctl status bluetooth
```
if not, you can start it
```
sudo systemctl start bluetooth
```

Let me know how it goes for you :)

evanescente-ondine, 2024-07-30:
Oh, huh, that's what it means... Sorry.
Works fine, I love not needing  those infinite kde or gnome dependencies !
Thanks a lot, plus TUI in general rock !!

pythops, 2024-07-30:
you're welcome, glad you like it :)

closed 2024-07-30 by pythops