pythops/bluetui #26: [Feature Request] Change order of adapter and devices

cvanelteren, 2024-10-08:
I find most my time just activating a connection which requires the additional tab press. Having the selected cell or even flipping the devices to the top would be greatly appreciated.

Thanks for the amazing tool!

pythops, 2024-10-08:
:thinking: not bad actually yeah

cvanelteren, 2024-10-24:
Not too familiar with rust, but browsing through the code I seems doable. There is no config file being use that you could specify the layout is there?

pythops, 2024-10-24:
I am working on it and I am taking the opportunity to do some refactoring as well. expect to be merged by the next week

closed 2024-11-25 by pythops

sermuns, 2025-09-15:
Great tool! I was wondering, how is this feature implemented? I cannot find it anywhere.