pythops/bluetui #28: Addition of Man Pages

299792458ms, 2024-11-12:
It would be useful to have man pages so to not check the repository documentation every time one forgets a keybind.

[image]

Github will not allow me to attach this file format, so I put it here, formatting and wording needs some work.

So here goes man page section 1:

```groff
.\" General Commands Manual.

.TH bluetui 1 "2024-07-29" "0.5.1" "General Commands Manual"
.SH NAME
bluetui \- manage bluetooth 

.SH SYNOPSIS
.B bluetui [OPTIONS]

.SH DESCRIPTION
.PP
TUI for managing bluetooth on Linux.

.SH OPTIONS
.PP
\f[B]-h, --help\f[R]
.PP
Print help
.PP
\f[B]-V, --version\f[R]
.PP
Print version

.SH DEFAULT BINDS
.SS GLOBAL
.PP
\f[B]s\f[R]
.PP
start scannning

.SS ADAPTER
.PP
\f[B]p\f[R]
.PP
toggle pairing

.PP
\f[B]o\f[R]
.PP
toggle power

.PP
\f[B]d\f[R]
.PP
toggle discovery

.SS PAIRED DEVICE
.PP
\f[B]u\f[R]
.PP
unpair device

.PP
\f[B]spacebar\f[R]
.PP
toggle connect

.PP
\f[B]t\f[R]
.PP
toggle trust

.PP
\f[B]e\f[R]
.PP
rename

.SH SEE ALSO
.PP
\fBbluetui(5)\fR

.SH BUGS
.PP
No known bugs.

.SH VERSION
.PP
0.5.1

.SH AUTHOR
.PP
Pythops <contact@pythops.com>, for more information see https://github.com/pythops/bluetui
```

and section 5:

```groff
.\" File Formats Manual.

.TH bluetui 5 "2024-07-29" "0.5.1" "File Formats Manual"

.SH NAME
.PP
bluetui \- configuration file

.SH DESCRIPTION
.PP
bluetui reads for keybinds in $HOME/.config/bluetui/config.toml. The syntax has to see the TOML format, see https://toml.io/en/ for more info.
.PP
If no configuration file is found then bluetui will default to the example configuration on this page.

.SH EXAMPLE
.PP
.nf
.RS 4
toggle_scanning = "s"

[adapter]
toggle_pairing = "p"
toggle_power = "o"
toggle_discovery = "d"

[paired_device]
unpair = "u"
toggle_connect = " "
toggle_trust = "t"
rename = "e"

[new_device]
pair = "p"

.SH SEE ALSO
.PP
\fBbluetui(1)\fR

.SH AUTHOR
.PP
Pythops <contact@pythops.com>, for more information see https://github.com/pythops/bluetui
```

Again, thanks for making impala and bluetui, I love it. And sorry for the formatting I just learned it now on he fly.

pythops, 2024-11-12:
you can type `?` to show help in popup and `esc` to discard it

closed 2024-11-12 by pythops

t-h2o, 2024-12-26:
@299792458ms, I love the idea of a manual page, my first reflex after to be lost with bluetui was to open a man page and then checkout the github.

You can easily write a man page with [asciidoc (man page)](https://docs.asciidoctor.org/asciidoctor/latest/manpage-backend/#doctype).

@pythops, why not have a manpage ?

pythops, 2024-12-26:
it's not that I am against manpage, I just don't see the value for it for a TUI that has builtin help with `?` keybinding. It's kind of expected now for TUIs to have a help section with `?`.