pythops/bluetui #30: Highlighted item is invisible with Solarized light theme

matejdro, 2024-11-20:
With solarized light theme, text of the highlighted item is invisible:

[image]

Theme:

[image]

pythops, 2024-12-15:
This PR https://github.com/pythops/bluetui/pull/32 should fix it
Can you plz check ?

mattkang, 2024-12-16:
@pythops was coming to report the same issue on `0.5.1`.  I was using everforest, and it was only an issue in the scanning pane for me.

**Fixed for me now, thanks!**
[image]
[image]

closed 2024-12-16 by pythops

matejdro, 2024-12-18:
Also can confirm it works from my side. Thanks a lot!

Any chance for a release with this fix?

pythops, 2024-12-18:
Hopefully by the end of the week or this weekend the latest

pythops, 2024-12-21:
@matejdro releaed now, enjoy

matejdro, 2025-01-02:
Can confirm, also fixed in release version. Thanks!

PR #32 (2024-12-16): fix light mode rendering fg
closes #30