pythops/bluetui #33: BUG: index out of bounds

mattkang, 2024-12-16:
```
thread 'main' panicked at /build/qmlvz394sva1rawai3r4xfjww5d0jpni-source/src/handler.rs:684:74:
index out of bounds: the len is 9 but the index is 10
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

This error occurred when I did the following:
1. Scan for devices
2. Stop scanning
3. Wait for some devices to fall off
4. Attempt to connect to one of the devices toward the end of the list

pythops, 2024-12-16:
What do you mean `Wait for some devices to fall off`
Can you explain the whole process more in details ?

mattkang, 2024-12-16:
@pythops edited the text. Happy to provide more detail as directed. It's my hunch that either devices are not intended to depopulate when scanning is stopped and/or this event is not being caught with indexing logic.

pythops, 2024-12-17:
I could reproduce it. Thanks for reporting it. I will look at it :eyes:

pythops, 2024-12-21:
To fix this one will require some refactoring and restructuring of the app, which I will do for sure. I will release a new version now cause some users are waiting for some changes and then I will focus on this interesting bug.

pythops, 2025-11-23:
fixed

commit 28834256 (2025-01-27): Add BackTab key handler

Fixes #33
closed 2025-11-23 by pythops