pythops/bluetui #39: Bluetui blank on startup, arch

patroldo, 2024-12-25:
Hi. For some reason when i start bluetui only blank screen appears

Version:
```
$ bluetui --version
bluetui 0.6.0
```

Bluetooth service is running
```
$ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/
systemd/system/bluetooth.service; enabled; preset: disabled
)
     Active: active (running) since Wed 2024-12-25 11:24:34 EET; 14min ago
 Invocation: 051ff7df69f24ed19721ccbe1464c1c1
       Docs: man:bluetoothd(8)
   Main PID: 636 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 38269)
     Memory: 3.3M (peak: 3.7M)
        CPU: 31ms
     CGroup: /system.slice/bluetooth.service
             └─636 /usr/lib/bluetooth/bluetoothd

Dec 25 11:24:34 archlinux bluetoothd[636]: src/plugin.c:
init_plugin() System does not support bap plugin
Dec 25 11:24:34 archlinux bluetoothd[636]: src/plugin.c:
init_plugin() System does not support bass plugin
Dec 25 11:24:34 archlinux bluetoothd[636]: src/plugin.c:
init_plugin() System does not support mcp plugin
Dec 25 11:24:34 archlinux bluetoothd[636]: src/plugin.c:
init_plugin() System does not support vcp plugin
Dec 25 11:24:34 archlinux bluetoothd[636]: profiles/audi
o/micp.c:micp_init() D-Bus experimental not enabled
Dec 25 11:24:34 archlinux bluetoothd[636]: src/plugin.c:
init_plugin() System does not support micp plugin
Dec 25 11:24:34 archlinux bluetoothd[636]: src/plugin.c:
init_plugin() System does not support ccp plugin
Dec 25 11:24:34 archlinux bluetoothd[636]: src/plugin.c:
init_plugin() System does not support csip plugin
Dec 25 11:24:34 archlinux bluetoothd[636]: src/plugin.c:
init_plugin() System does not support asha plugin
Dec 25 11:24:34 archlinux bluetoothd[636]: Bluetooth management interface 1.23 initialized
```

pythops, 2024-12-25:
:thinking: 
Does `bluetoothctl` command works for you ?

patroldo, 2024-12-25:
Hi, @pythops 
Next time when i reproduce this problem - i will let you know

Basically you propose to connect to bluetooth via bluetoothctl and see if it works, right?

pythops, 2024-12-25:
I just try to know where the problem is for you, if `bluetoothctl` works, then `bluetui`  should work as well. But I suspect something with your setup,  with all the log messages you showed above specially this one
```
o/micp.c:micp_init() D-Bus experimental not enabled
```
`bluetui` (same as bluetoothctl) uses `dbus` to interact with the daemon, so if something is off with `dbus` then it won't work.

patroldo, 2024-12-25:
[yurii@archlinux ~]$ bluetoothctl 
[bluetooth]# Agent registered
[bluetooth]#

Just reproduced. Looks like bluetoothctl is working fine

patroldo, 2024-12-25:
Hm. Just checked dmesg and this is what i see
```
$ sudo dmesg | grep -i bluetooth
[sudo] password for yurii: 
[    8.954235] Bluetooth: Core ver 2.22
[    8.954257] NET: Registered PF_BLUETOOTH protocol family
[    8.954259] Bluetooth: HCI device and connection manager initialized
[    8.954263] Bluetooth: HCI socket layer initialized
[    8.954266] Bluetooth: L2CAP socket layer initialized
[    8.954270] Bluetooth: SCO socket layer initialized
[    8.983038] Bluetooth: hci0: CSR: Setting up dongle with HCI ver=9 rev=0001
[    8.983042] Bluetooth: hci0: LMP ver=9 subver=0001; manufacturer=2279
[    8.983044] Bluetooth: hci0: CSR: Unbranded CSR clone detected; adding workarounds and force-suspending once...
[    8.983048] Bluetooth: hci0: CSR: Couldn't suspend the device for our Barrot 8041a02 receive-issue workaround
[    8.983053] Bluetooth: hci0: HCI Delete Stored Link Key command is advertised, but not supported.
[    8.983055] Bluetooth: hci0: HCI Read Default Erroneous Data Reporting command is advertised, but not supported.
[    8.983057] Bluetooth: hci0: HCI Set Event Filter command not supported.
[    9.880170] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    9.880174] Bluetooth: BNEP filters: protocol multicast
[    9.880178] Bluetooth: BNEP socket layer initialized
[   12.166963] Bluetooth: hci0: Opcode 0x1005 failed: -110
[   12.166969] Bluetooth: hci0: command 0x1005 tx timeout
```

Looks like issue with drivers, device or configuration  i guess, so it's not a problem directly connected to bluetui. But maybe it's possible to add logs. Or if it's already there - would be helpful to add

patroldo, 2024-12-25:
After checking arch linux forum - looks like i have bluetooth dongle which is not recognized as known by linux kernel. Basic w/a which just worked for me from this post - https://bbs.archlinux.org/viewtopic.php?id=274953. Steps:

1) create this file:
/etc/modprobe.d/btusb.conf

2) Add following line:
options btusb enable_autosuspend=0

So i hope problem is gone. For now logs from dmesg looks like this:
```
$ sudo dmesg | grep -i bluetooth
[    7.636781] Bluetooth: Core ver 2.22
[    7.636810] NET: Registered PF_BLUETOOTH protocol family
[    7.636812] Bluetooth: HCI device and connection manager initialized
[    7.636818] Bluetooth: HCI socket layer initialized
[    7.636821] Bluetooth: L2CAP socket layer initialized
[    7.636827] Bluetooth: SCO socket layer initialized
[    7.702296] Bluetooth: hci0: CSR: Setting up dongle with HCI ver=10 rev=0089
[    7.702300] Bluetooth: hci0: LMP ver=10 subver=0089; manufacturer=2279
[    7.702303] Bluetooth: hci0: CSR: Unbranded CSR clone detected; adding workarounds and force-suspending once...
[    7.702307] Bluetooth: hci0: CSR: Couldn't suspend the device for our Barrot 8041a02 receive-issue workaround
[    7.702309] Bluetooth: hci0: HCI Delete Stored Link Key command is advertised, but not supported.
[    7.702312] Bluetooth: hci0: HCI Read Default Erroneous Data Reporting command is advertised, but not supported.
[    7.702313] Bluetooth: hci0: HCI Set Event Filter command not supported.
[    8.549860] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    8.549865] Bluetooth: BNEP filters: protocol multicast
[    8.549871] Bluetooth: BNEP socket layer initialized
[    8.941422] Bluetooth: MGMT ver 1.23
[   14.028745] Bluetooth: RFCOMM TTY layer initialized
[   14.028760] Bluetooth: RFCOMM socket layer initialized
[   14.028766] Bluetooth: RFCOMM ver 1.11
```

And bluetooth service logs:
```
$ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/
systemd/system/bluetooth.service; enabled; preset: disabled
)
     Active: active (running) since Wed 2024-12-25 15:12:32 EET; 4min 17s ago
 Invocation: b45b6736510847689885774696a80125
       Docs: man:bluetoothd(8)
   Main PID: 625 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 38269)
     Memory: 3.6M (peak: 4.3M)
        CPU: 73ms
     CGroup: /system.slice/bluetooth.service
             └─625 /usr/lib/bluetooth/bluetoothd

Dec 25 15:12:38 archlinux bluetoothd[625]: Endpoint registered: sender=:1.27 path=/MediaEndp
oint/A2DPSource/aptx_ll_duplex_0
Dec 25 15:12:38 archlinux bluetoothd[625]: Endpoint registered: sender=:1.27 path=/MediaEndp
oint/A2DPSource/faststream
Dec 25 15:12:38 archlinux bluetoothd[625]: Endpoint registered: sender=:1.27 path=/MediaEndp
oint/A2DPSource/faststream_duplex
Dec 25 15:12:38 archlinux bluetoothd[625]: Endpoint registered: sender=:1.27 path=/MediaEndp
oint/A2DPSink/opus_05
Dec 25 15:12:38 archlinux bluetoothd[625]: Endpoint registered: sender=:1.27 path=/MediaEndp
oint/A2DPSource/opus_05
Dec 25 15:12:38 archlinux bluetoothd[625]: Endpoint registered: sender=:1.27 path=/MediaEndp
oint/A2DPSink/opus_05_duplex
Dec 25 15:12:38 archlinux bluetoothd[625]: Endpoint registered: sender=:1.27 path=/MediaEndp
oint/A2DPSource/opus_05_duplex
Dec 25 15:13:06 archlinux bluetoothd[625]: No matching connection fo
r device
Dec 25 15:13:19 archlinux bluetoothd[625]: profiles/audio/avctp.c:av
ctp_control_confirm() Control: Refusing unexpected connect
Dec 25 15:13:55 archlinux bluetoothd[625]: /org/bluez/hci0/dev_68_59_32_87_39_1C/sep2/fd1: f
d(31) ready
```

pythops, 2024-12-25:
Good point, I will see if I can print logs to stderr