pythops/bluetui #42: feature: pin authentication

dominik-muc, 2025-01-10:
Hi! Today I wanted to use your tool to connect to my BT keyboard. Unfortunately the app didn't show the PIN I had to enter on my KB, so I used `bluetoothctl`. This would be a very nice feature to have!

pythops, 2025-01-10:
:thinking: it is supported normally. A popup should appear with the PIN  
Would you mind share a screenshot about the popup that it shows you ?

dominik-muc, 2025-01-10:
Its hard to take a screenshot because the pop-up disappears very fast. It was a pop-up in right top corner that said „bluetooth: authentication cancelled”.
When I was connecting my phone (request from my phone) to the PC it correctly showed an authorization prompt asking if the digits match. The issue was when I initiated pairing from my PC to the keyboard, for a while it wouldn't do anything (about 1 minute) and then the error popped up, probably due to internal timeout.

I'm using Arch Linux + sway and alacritty as terminal.

pythops, 2025-05-10:
The popup on the right top is for notification and it is suppose to disappear after couple of seconds. Yes there is a timeout for the authentication before it is canceled. 
As this feat is already working I am gonna close this issue

closed 2025-05-10 by pythops