pythops/bluetui #43: Selected row not (visibly) highlighted with Solarized Dark colorscheme

stefankreutz, 2025-01-12:
I can't see which row is selected when using the [Solazired](https://ethanschoonover.com/solarized/) Dark colorscheme because the [`DarkGray`](https://docs.rs/ratatui/latest/ratatui/style/enum.Color.html#variant.DarkGray) highlight background is indistinguishable from the Solarized background color.

There was a similar issue #30 with the Solarized Light colorscheme, which was fixed in PR #32.

I would like to see a highlight symbol such as a leading arrow because that should work for everybody, regardless of the colorscheme. I'll submit PR shortly.

pythops, 2025-01-12:
Can you share a screenshot ?

stefankreutz, 2025-01-12:
Here is a screenshot. The first paired device should be highlighted.

[image]

pythops, 2025-01-12:
yeah I can add a highlight symbol for the selected device like the adapter section

stefankreutz, 2025-01-12:
That would be great!

Here is a corresponding PR #44. Feel free to merge or discard.

pythops, 2025-01-12:
hahahah we just made it at the same time
https://github.com/pythops/bluetui/pull/45

pythops, 2025-01-12:
I added it to the other section as well. If you don't mind we can keep my PR :)

stefankreutz, 2025-01-12:
Of course. Thank you!

closed 2025-01-12 by stefankreutz

pythops, 2025-01-12:
if you can approve it then I will merge it :)

stefankreutz, 2025-01-12:
I've tested I locally and am happy with it. Is there any box I should tick to approve? :)

pythops, 2025-01-12:
no need, merged, enjoy :)