pythops/bluetui #52: Couldn't install on Raspberry Pi Zero 2 W and RaspianOS Lite

dbrunker, 2025-02-26:
I don't know Rust.  I wasn't able to find a pre-compiled binary of bluetui for RaspianOS Lite on a Raspberry Pi Zero 2 W.  I used:

`cargo install bluetui`

I'm not sure if I was supposed to use sudo or not but sudo wasn't listed in the instructions.  I went to bed and in the morning my screen said:

```
[...other compiled names above...]
   Compiling dbus-crossroads v0.5.2
   Compiling tokio-stream v0.1.17
   Compiling dbus-tokio v0.7.6
   Compiling bluetui v0.6.0
error: could not compile `bluer` (lib)

Caused by:
  process didn't exit successfully: `/home/######/.rustup/toolchains/stable-aarch64-unknown-linux-gnu/bin/rustc --crate-name bluer --edition=2021 /home/######/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/bluer-0.17.3/src/lib.rs --er
ror-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --diagnostic-width=240 --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C linker-plugin-lto -C codegen-units=1 --cfg 'feature="bluetoothd"' --cfg 'featu
re="custom_debug"' --cfg 'feature="dbus"' --cfg 'feature="dbus-crossroads"' --cfg 'feature="dbus-tokio"' --cfg 'feature="default"' --cfg 'feature="displaydoc"' --cfg 'feature="full"' --cfg 'feature="id"' --cfg 'feature="l2cap"' --cfg 'featu
re="lazy_static"' --cfg 'feature="mesh"' --cfg 'feature="pin-project"' --cfg 'feature="rfcomm"' --cfg 'feature="serde"' --cfg 'feature="tokio-stream"' --check-cfg 'cfg(docsrs,test)' --check-cfg 'cfg(feature, values("bluetoothd", "custom_deb
ug", "dbus", "dbus-crossroads", "dbus-tokio", "default", "displaydoc", "full", "id", "l2cap", "lazy_static", "mesh", "pin-project", "rfcomm", "serde", "tokio-stream"))' -C metadata=1acb4394b2b1dbfc -C extra-filename=-bf1672ef88f7c641 --out-
dir /tmp/cargo-installuAeQB8/release/deps -C strip=symbols -L dependency=/tmp/cargo-installuAeQB8/release/deps --extern custom_debug=/tmp/cargo-installuAeQB8/release/deps/libcustom_debug-8a7241156431192c.rmeta --extern dbus=/tmp/cargo-insta
lluAeQB8/release/deps/libdbus-2a4fde1e7df3316a.rmeta --extern dbus_crossroads=/tmp/cargo-installuAeQB8/release/deps/libdbus_crossroads-578fe75915471fa8.rmeta --extern dbus_tokio=/tmp/cargo-installuAeQB8/release/deps/libdbus_tokio-3177c845ea
ad4356.rmeta --extern displaydoc=/tmp/cargo-installuAeQB8/release/deps/libdisplaydoc-fb97f382c0fc62ec.so --extern futures=/tmp/cargo-installuAeQB8/release/deps/libfutures-61ec48cf0b689bb8.rmeta --extern hex=/tmp/cargo-installuAeQB8/release/
deps/libhex-c1c01142d601527e.rmeta --extern lazy_static=/tmp/cargo-installuAeQB8/release/deps/liblazy_static-eee1a5924ca7ddfb.rmeta --extern libc=/tmp/cargo-installuAeQB8/release/deps/liblibc-0fbd8b8ffb496b7a.rmeta --extern log=/tmp/cargo-i
nstalluAeQB8/release/deps/liblog-04ce0f2cd42f591e.rmeta --extern macaddr=/tmp/cargo-installuAeQB8/release/deps/libmacaddr-dc02fc6ef82321cb.rmeta --extern nix=/tmp/cargo-installuAeQB8/release/deps/libnix-91b88e20f26b87ae.rmeta --extern num_d
erive=/tmp/cargo-installuAeQB8/release/deps/libnum_derive-b8a4b26e3e10e3c2.so --extern num_traits=/tmp/cargo-installuAeQB8/release/deps/libnum_traits-1146481274b8960d.rmeta --extern pin_project=/tmp/cargo-installuAeQB8/release/deps/libpin_p
roject-45b6a161ed2fcb3c.rmeta --extern serde=/tmp/cargo-installuAeQB8/release/deps/libserde-604035227da4d51d.rmeta --extern strum=/tmp/cargo-installuAeQB8/release/deps/libstrum-a04c572ac4f7f04b.rmeta --extern tokio=/tmp/cargo-installuAeQB8/
release/deps/libtokio-b26335269d5d61d0.rmeta --extern tokio_stream=/tmp/cargo-installuAeQB8/release/deps/libtokio_stream-99832325d9e69806.rmeta --extern uuid=/tmp/cargo-installuAeQB8/release/deps/libuuid-0be753d1ecc2c996.rmeta --cap-lints a
llow -L native=/usr/lib/aarch64-linux-gnu` (signal: 9, SIGKILL: kill)
error: failed to compile `bluetui v0.6.0`, intermediate artifacts can be found at `/tmp/cargo-installuAeQB8`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
######@raspberrypi:~ $sudo screendump 1 > error.txt
```

I'm sorry I don't know how to fix this myself.  I'd be grateful for any help

pythops, 2025-02-27:
Have you tried this binary from the release page `https://github.com/pythops/bluetui/releases/download/v0.6/bluetui-aarch64-linux-gnu` ?

dbrunker, 2025-02-28:
Sorry I just noticed this on neofetch:
OS: Debian GNU/Linux 12 (bookworm) **aarch64**

pythops, 2025-02-28:
did you give it a try first ? just follow the instructions on Readme and it should work

closed 2025-02-28 by pythops

dbrunker, 2025-02-28:
Yes, it worked.  Thank you, very much, pythops.  I couldn't have done it without you.