pythops/bluetui #64: New bluetui logo proposal

Bugg4, 2025-07-19:
Hi! Had some fun designing a new logo for the bluetui README:

[image]

This is a mockup, the actual file will be a vector svg.
I intend to make it follow GitHub's theme dynamically by implementing the method I found in this [blog post](https://paul.af/github-readme-dark-mode)

Let me know if you like it, and I'll submit a PR :)

pythops, 2025-07-20:
Looks Great 🔥 
my favorite is this one

[image]

I am wondering it is possible to make `L` continue ?

pythops, 2025-07-20:
Yeah I would prefer the monochrome.
 I like the idea of the cursor. if it is was at the beginning or at the end maybe it would be more clear to see, what do you think ?

Bugg4, 2025-07-20:
How's this ?
[image]

Bugg4, 2025-07-21:
Glad you like it!
My intention was to make it automatically adapt to Github's theme through the method I linked to in the first comment.

So the background would be transparent, and the logo would adapt to be white or black based on the theme, that's why I showed you both variants.

But if you want it to be always white on a black background, we can leave it static 👍🏻

closed 2025-07-22 by pythops

pythops, 2025-07-22:
@Bugg4 I have other TUIs that need logos as well if you want 😄

Bugg4, 2025-07-22:
Sure! I'll take a look 👀